free-dmo/free-dmo-stm32 #1: CRC algorithm

garci66, 2022-03-29:
Not sure if it helps (and sorry for opening an issue). But in the past I've used this tool to brute force unknown CRC checksums and find the coefficients. https://github.com/nitram2342/bruteforce-crc

Means we probably need a few different samples, but I've used it to decode a protocol with its own CRC and this tool helped me find the proper coefficients. Hope it helps !!!

freepdk, 2022-03-29:
I tried to brute force the poly already. No luck for me.

freepdk, 2022-03-29:
In case you want to try yourself, here one data block (common in all read outs):
```
static const uint8_t crcblk1[] = {
  0x00,0x00,0x00,0x00, 0x00,0x00,0x00,0x00, 0x00,0x00,0x00,0x00, 0x00,0x00,0x00,0x00, 
  0x00,0x00,0x00,0x00, 0x00,0x00,0x00,0x00, 0x00,0x00,0x00,0x00, 0x00,0x00,0x00,0x00, 
  0x00,0x00,0x00,0x00, 0x00,0x00,0x00,0x00,
};
```
CRC for this is: 0x913ec3dd (or 0xddc33e9 - depending on byte order).

Another one to verify:
```
static const uint8_t crcblk2[] = {
  0x01,0x00,0x00,0x00, 0x00,0x00,0x00,0x00, 0x00,0x00,0x00,0x00, 0x00,0x00,0x85,0x30,
  0xE1,0x07,0x00,0x00, 0x00,0x00,0x00,0x00,
};
```
CRC for this is: 0x8723b272 (or 0x72b22387 - depending on byte order).

garci66, 2022-03-29:
Thanks. Let me see

kategray, 2022-03-29:
Doesn't look like a standard CRC32 algorithm, if the input is correct.

https://crccalc.com/?crc=00000000000000000000000000000000000000000000000000000000000000000000000000000000&method=crc32&datatype=hex&outtype=0

dtomcat, 2022-04-13:
If your question/issue is resolved, please close the issue so attention can be focused on open issues.

eychei, 2022-07-18:
Looking into this right now.
How did you find out the block size for the crc?
Do you have a memory map of the Tag data which you identified already?