free-dmo/free-dmo-stm32 #5: How Accomplished

0Kills, 2022-04-04:
This isn't an issue, but are you able to detail how you reverse engineered the protocol/tag format and how it works? I cracked open a 550 and tried to get started reversing the pins between the board and the daughterboard, but I got lost in the process. How did you determine it was I2C, how were you able to reverse the protocol, and how does the protocol work?  Does the digital signature not also sign the blocks that hold the label count? Seems almost pointless for them to use a digital signature when it can just be copied byte for byte.

freepdk, 2022-04-05:
First you identify the ICs on the main board.

==> You learn it is a STM32F072

Then you look at the pinout of the STM32F072 and identify which pins go to the connector for the RFID board.

==> here you learn they are 2 x I2C pins, 1 x GPIO pin, 1 x 3V3, 1 x GND, and 1 pin unknown (this we identify later when looking at the RFID board)

Now you look at the RFID board and identifiy the RFID IC from NXP: CLRC663

==> Again you look at the data sheet and trace the pins ==> you confirm I2C, 3V3 and GND. Then you identify the GPIO pin is connected to IRQ and the unknown pin is in fact PowerDown.

Since you now know all the pins from the cable between main board and RFID board you attach a logic analyzer (e.g. Saleae Logic 8 - or any cheap logic analyzer you can find).

Usually the logic analyzer is having some software which you can use to decode the data (e.g. I2C).

==> You now have the byte stream going in and out via I2C (see here: https://www.eevblog.com/forum/reviews/dymo-550-thermal-printer-drm-hacking/msg4047715/#msg4047715)

Now again you read the CLRC663 data sheet to understand the data going in and out.

==> You find the commands which is just set up the CLRC633 and also the data which is sent to the real tag presented to the reader and also the response from the tag.

Now you look up the data sheet for the tag: NXP SLIX2  and find the meaning of the data sent:

- read UID
- read signature
- read data blocks
- write counter
...

All what is left is to capture the data which is sent / received from real tag and write an emulation to play back this data.

freepdk, 2022-04-05:
> Does the digital signature not also sign the blocks that hold the label count? Seems almost pointless for them to use a digital signature when it can just be copied byte for byte.

The signature is in the SLIX2 specification which just signs the UID (8 byte ID) of the tag. 
D.MO at least uses their own signature (own ECC priv key to sign in factory, pub key to verify in firmware), but like you mentioned this does not protect from replay. 

==> All in all THIS WAS NOT DESIGNED TO BY SECURE (no intention visible from D.MO).

Like I mentioned before. They used the wrong security scheme... maybe they had to save money :-P

0Kills, 2022-04-05:
Couldn't they conceivably have the private key in a secure enclave inside the device to digitally sign new data blocks after decrement without exposing the key? 
Also I've just dumped the data from the 25 x 54mm labels. What is the best way to get to you to add new label emulation?

freepdk, 2022-04-06:
They could have done many things differently... But I see no point in discussing ideas here which would help them or others to make a "better" system. The complete idea is WRONG!

You can attach the dump .txt file here (When you comment here, there is a small text below the edit box "Attach files by dragging...")

0Kills, 2022-04-06:
[E0040108449690ED_20220405143943.txt](https://github.com/free-dmo/free-dmo-stm32/files/8426840/E0040108449690ED_20220405143943.txt)

Good point about not helping them. Label dump for 25x54mm is attached. Thanks for all your hard work.

freepdk, 2022-04-06:
Thanks for the dump. Intgerated with commit: 89fa322

dtomcat, 2022-04-13:
If your question/issue is resolved, please close the issue so attention can be focused on open issues.

RyanEwen, 2023-09-27:
This is great. I have a Zebra ZSB-DP14 thermal label printer which uses refills equipped with tiny little PCBs, just big enough to house 3 pads that align with 3 pins in the printer. I'm hoping to figure out what is communicated between the refill PCBs and the printer so that I'm not locked into their expensive refills.

hinakhan09, 2024-07-11:
Download Now https://pclife.info/dymo-connect-crack/