free-dmo/free-dmo-stm32 #9: Ready bins

netboy69, 2022-04-12:
Hi, can we please have all bins for all kind of labels? So we dont have to compile or use the reader from the printer? Regards

dtomcat, 2022-04-13:
IMHO if you’re unable to compile… you should probably not do this mod.   The creator has given you everything you need. It would be a lot of work to keep updating bins with each possibility

netboy69, 2022-04-13:
I'm able to compile but just didn't work, if i try the bin that is already done works perfect, but that's not the size of labels that i use. That's why I'm was just asking if anyone can do it. But thanks anyway I'll find a way. Kind regards

dtomcat, 2022-04-13:
What issues are you having? Perhaps the community can help. My blue pills arrive tomorrow so I’ll be attempting everything tomorrow.  The bin is created?  You program the chip and it doesn’t work?  Checked all connections?

netboy69, 2022-04-13:
I got the blue pill and the programer for the pill, but i was missing the 4.7om resisters , so i thought just to use the pill to try, I've change the code to match the label i wanted, compile all under Ubuntu got the final bin, content the pill to printer and wouldn't detect the label size, with the original bin already compiled the printer assume a label no problem. Today i should receive the 4.7 om resisters so I'll build the all thing. And see. Regards

netboy69, 2022-04-13:
Also I'm trying on a 5xl because the original bin worked fine I'll assume all should work 100%, I'll post my result soon and if i manage to compile the bins correctly I'll post them all here it's just quicker, compiling takes time when you do it first time. Regards

freepdk, 2022-04-13:
I created a github workflow which automatically compiles all variants for new releases. 
Have a look at the main page / releases section where you can download a ZIP file containing firmwares for all default emulation variants.

netboy69, 2022-04-13:
Omg you are the man, thanks alot, can't wait to get home and program the blue pill. I wanted to do that but I dont know why my bin didn't work because Ubuntu show no errors. One more time many thanks for that. Have a great day.

freepdk, 2022-04-13:
Unfortunately (or fortunately?) I do not have a 5XL.

Since the printer has a different main MCU it might be that they have differences in their firmware implementation which might need some tweaks to the emulation.

However you reported your 5XL with the pre compiled firmware worked already :-)

One thing to note, after changing the firmware on the blue pill, make sure to power cycle the printer completely (remove USB and POWER for 5 seconds). Mine gets upset when I change the blue pill firmware while the printer is running (I use the SWD pins connected to ST-Link, so I can use debugger).

netboy69, 2022-04-13:
By the end of the day I'll post all results, but yes the 5xl worked fine with the default label on the freedom.bin, I'll report the rest later on. Kind regards

netboy69, 2022-04-13:
Hi every one, so here my results on a Dymo 5XL, first tests were made with the STM32 blue pill only and not connected to the reader, i have tested all bins files and the results were: freedmo-default-sku-S0722400_36x89mm.bin, freedmo-default-sku-S0722430_54x101mm.bin and freedmo-default-sku-S0722550_19x51mm.bin the printer detects the label and setups it self for the label size on the other bins files it does not detect the label size meaning if you press the right button to skip a or print blank label because the size is wrong it wont skip the blank label, but if you tell the printer to print a 1744907 for example it will print fine and stops ok on the next label and gives no errors. so it can be used like this just remember to setup the size of the bin you programed and will get the job done. Then my next test was to connect to the rfid reader all seem fine but i dont have original label to test if it reads fine and adjust the size, ill see if i can get genuine labels and come back for a result, but so far with the stm32 blue pill alone works fine you dont need to connect to the reader if you only print one kind of labels.

netboy69, 2022-04-13:
No worries, in the end my bin was the same because both had the same result. I just thought I was doing something wrong because it wasn't working like the original bin. In the end i got the 5xl printing on the 6x4 labels bypassing the RFID so happy days and no money for dymo genuine labels. :-)

dtomcat, 2022-04-14:
My update:  Programmed one Blue Pill into an ST-Link V2...   compiled src (src prior to update with github workflow) to use 4x6 labels.  uploaded resulting BIN to a new Blue Pill (via previously mentioned blue pill) and everything works fine. (I think)  I only have 4x6 labels, so I can't tell if it's detecting my labels or not.  Will this auto top off my labels to 50 if it's below that?  and in the src it says 220 labels, but defaults to 50.  Other than that... seems to work.  I may pick up a pack of other labels to see if it recognizes those or not.

So this confirms that it works fully with 5XL if issues noted above are normal.  
Thanks for your hard work on this!

dtomcat, 2022-04-14:
so instead of buying new labels, i tried flashing a different firmware... none worked... either it came up 4x6 (no matter which firmware) or only worked with original labels... reflashed the bin I compiled and now it shows 220 labels no matter which label i insert. (4x6)...  the default... if i load generic labels... it shows 220 labels (as expected)... if i load my roll, it shows 220 which rfid tag says there is 50 labels.   

Should it show tag amount if genuine labels?  i can't tell if it's reading the labels or not.

netboy69, 2022-04-14:
On the 4x6 labels it won't detect the label but let say you are painting a pdf on the setting chose the correct label size and it will print. I assume by not drying the label size will also don't know how many labels left.

netboy69, 2022-04-14:
Printing not paiting

netboy69, 2022-04-14:
Windows, go to printers then chose the 5xl then settings and chose correct label size.

netboy69, 2022-04-14:
whats new on the firmware 1.0.5?

dtomcat, 2022-04-14:
Looks like he added 1”x1” label to code

netboy69, 2022-04-14:
Ho ok, I'll try when possible. Thanks

dtomcat, 2022-04-15:
4x6 is the only one that I actually got working.  And i figured out, if you have real dymo labels inserted, then every label inserted after keeps the orginal inserted.  i.e. I insert my 4x6 label (real) with 50 remaining... then insert generic label....   it shows up as 4x6 with 50 remaining.   if I power down and power back up... that same roll shows up as 4x6 with 220 labels...   If I insert the real labels... they also show up with 220 remaining.    all the other pre-compiled bins... would not show any labels ... only real labels... not sure whats wrong... but i'll only be using 4x6 labels in the 5xl... so i'm not worried...    i have it working like i need it to.   I have a different size label coming from Amazon to try out.   I'll follow up with the results.

netboy69, 2022-04-15:
Nice, it's like me i only use the 4x6 and it's printing ok so far so let's see if this issue get fixed next updates about detecting the labels. But I'm happy so far because I can use 3rd party labels.

dtomcat, 2022-04-17:
Tried new Labels... after power cycle... Worked great.  Right labels/remain amount showed...  

Loaded 4x6 labels... and showed as new labels...  so even if using real labels, you have to power cycle after each change.  Whatever is loaded upon power up... is the only label that shows until another power cycle.

netboy69, 2022-04-17:
Perfect we can live with that.

freepdk, 2022-04-18:
> Tried new Labels... after power cycle... Worked great. Right labels/remain amount showed...
>
> Loaded 4x6 labels... and showed as new labels... so even if using real labels, you have to power cycle [...]

Hmmm... maybe we can improve on that. On my 550 when you remove a spool (so RFID board can't see it anymore), wait 3 seconds and put in a new spool it shows the new labels with a fresh maximum counter.

Maybe the 5XL needs some other timing... Anyone with a STLinkV2 connected to the bluepill could debug on 5XL?

dtomcat, 2022-04-18:
> Hmmm... maybe we can improve on that. On my 550 when you remove a spool (so RFID board can't see it anymore), wait 3 seconds and put in a new spool it shows the new labels with a fresh maximum [...]

I can probably help.  Just let me know what to do.  I have a home made ST-Link V2… but should work

dtomcat, 2022-04-20:
I’ll be creating another D.MO today for testing (longer leads). Let me know what you need

dtomcat, 2022-04-20:
Test Blue Pill is ready...  not sure how to get the information you need, so waiting on your response.

dtomcat, 2022-04-21:
I could pull the pin out of the connector instead of desoldering.  (To save time and headache) Just let me know whenever you have time and I’ll do my part

dtomcat, 2022-05-04:
Any update on what you need me to do to get you the data you need?  Do I need a special build?

dtomcat, 2022-07-31:
?????  still supporting this?  if so, I'd like to help troubleshoot the 5xl

dtomcat, 2022-08-03:
Ah. Sorry to hear this

dtomcat, 2022-08-03:
Egg grill

dtomcat, 2022-10-25:
Just checking in to see if I can help?  (Troubleshoot the 5xl