free-dmo/free-dmo-stm32 #13: 6 pin connector

0Kills, 2022-04-21:
The 6 pin connector recommended in the guide does not seem to match the connector on the board. 
The one on the board has notches in the front (or back) while this cable has them on the sides. This is a JST ZH 6 pin cable from Amazon: https://www.amazon.com/gp/product/B07Q5F9CCZ/ref=ppx_yo_dt_b_asin_title_o07_s00?ie=UTF8&psc=1
[image]

dtomcat, 2022-04-21:
This is what I used:
1.25mm Pitch Connectors and Pre-Crimped Cables Compatible with Molex PicoBlade for Pixhawk Silicone 15cm Wire https://www.amazon.com/dp/B07S18D3RN/ref=cm_sw_r_cp_api_i_XMXH33ZB033DANEVMGSC?_encoding=UTF8&psc=1

freepdk, 2022-04-21:
Actually I used the listing title from ebay which was:  "6pin 1.25mm Wire To Board Connector / JST ZH 6pin"

So maybe the listing I purchased from was wrong or the listing you found on amazon is wrong...

0Kills, 2022-04-28:
The link from @dtomcat worked for me.

GigaElectron, 2022-08-26:
@freepdk JST ZH has a 1.5mm pitch, whereas JST **_GH_** has a 1.25mm pitch, the listing you used must have been incorrect. The `README` should be changed to say "JST GH". 👍

freepdk, 2022-08-28:
I changed the README to say GH (see commit 02bf8f9)

soerendohmen, 2023-06-23:
GH is definitely not the correct connector, as you can see on the picture on JST's website: [https://jst.de/product-family/show/89/gh](https://jst.de/product-family/show/89/gh)

queengooborg, 2023-09-15:
It appears that the GH connector is commonly confused with the Molex PicoBlade (MX) connector.  They're both the same pitch (1.25mm).  I've updated the readme on my fork (https://github.com/queengooborg/free-dmo-stm32#required-parts) and will be submitting a PR to help correct this info upstream!