free-dmo/free-dmo-stm32 #16: How to debug for message: Unknown label detected

malebuffy, 2022-05-28:
Hi guys,

I received my bluepill with what it seams a STM32F103C8x ship on it. At least it looks legitimate. I tried both flashing the bin files and also compiling anf flashing the freedmo.bin. Wire connections are OK, I am using the 4.7k resistors as well in order to be able to recognize different spools. 

After flashing, I connect the bluepill to the printer and when I switchit on it says: Unknown Label Detected

No matter which Bin I flash, same thing. Do you have any tips for me how to debug? I am getting kind of desprate...

Thanks for any feedback

malebuffy, 2022-05-29:
Checked all the pins, checked connections. All good. Can it be that because I am using the latest Dymo Software v1.4.3.131 that they blocked the UIDs? Anyone has an older version I could try?

dtomcat, 2022-05-31:
[image]

also ensure jumpers are right

dtomcat, 2022-05-31:
[image]

dtomcat, 2022-05-31:
[image]

malebuffy, 2022-05-31:
Oh sorry, silly me! Yes, the jumpers are like this. I never touched them.

malebuffy, 2022-05-31:
No problem. So does it work on your 5XL? Do you use an original STM32 or some clone?

dtomcat, 2022-05-31:
[image]

This is the change for me.  Without troubleshooting, this is how it works for me… but is ok for me as I don’t change labels frequently

malebuffy, 2022-05-31:
I will try to build your fork and report back. BTW do you use original STM32 or clone?

dtomcat, 2022-05-31:
My fork is basically the same… just added a new label. And updated the readme.  

As far as I know… it’s original. But cloners are good!

malebuffy, 2022-05-31:
Nothings works. I am starting to think that I have a fake STM and somehow the peripherals are not correctly cloned. I removed the resistors and disconnected the RFID part, still receiving an error. Sadly not easy to debug like an arduino for exampl3 where you just read from serial monitor.

I have ordered a new one to see if I have better results.

malebuffy, 2022-06-01:
Fixed it! It was a bad board. I used another one and remade the connections. It works now without the resistors and the RFID board, but I will try that in the evening and report back!

malebuffy, 2022-06-01:
I did it. Board was the culprit. Bad luck I guess. I can confirm the same behaviour when I have an original roll in. I have to power cycle.

malebuffy, 2022-06-01:
Thanks to dtomcat for the support and to the author for the mod.

dtomcat, 2022-06-01:
No worries. I figured it had to be a bad board if everything else i mentioned was good. Glad you got it working.  And good to know yours functions same as mine.  So must be a timing issue (difference between 550 and 5XL)