free-dmo/free-dmo-stm32 #20: power reset issue

jody-steele, 2022-09-15:
i am having an issue with the 550 seeing labels as supported after a power off and back on. to get it back to seeing supported labels i have to click the reset button on the stm32. otherwise its working ok. maybe a flash issue but I have tried re-flashing and full erase but stil running into the issue. 

also you listed 4.7k ohm res but I couldn't tell if they are half 1/2w or what. can you add more info on which 4.7k Ohm to use.

jody-steele, 2022-09-15:
is there anything I need to do for programming the stm32 so that on power on its ready to go. I am using the 30256 bin file if that matters.

jody-steele, 2022-09-29:
Any help would be great. every time the device powers on i have to hit the reset button the bluepill and then it works great. at least until the next power off and on.

freepdk, 2022-10-09:
Hi,

It's hard to diagnose your problem. 

The resistor just needs to be 4.7k with almost no wattage requirements. It is just a pull up. So really any 4.7k resistor you can find will work.

If your bluepill has startup problems you should check if your have a genuine STM32F103 or a clone. Sometimes the clones do have different behavior which might or might not be tweeked with firmware or hardware modifications.

jody-steele, 2022-10-09:
Issue is that a reset resolves the problem. So, I built the reset button onto the back of the printer. If I send a link to the Amazon item could u check it and see if it looks right?

Thank you

Sent via the Samsung Galaxy S21 Ultra 5G, an AT&T 5G smartphone

freepdk, 2022-10-09:
A picture of your actual bluepill (most interesting the labeling on the main / biggest IC) would reveal the truth.

Here is a very nice writeup from keirf:

https://github.com/keirf/Greaseweazle/wiki/STM32-Fakes

The first picture is a "good one", all following ones are fakes with problems.