free-dmo/free-dmo-stm32 #22: Tag Data

lightkeepr, 2022-11-06:
How do I send you the tag data I read off of a 30270 Receipt paper roll?

freepdk, 2022-12-01:
You can attach the file here (there is a "Attach files by dragging..." on the bottom of the edit box when you add a comment.

lightkeepr, 2022-12-06:
[E004010851A29555_20221106152105.txt](https://github.com/free-dmo/free-dmo-stm32/files/10161097/E004010851A29555_20221106152105.txt)

Should be attached!

czester1994, 2023-08-03:
[E004010849D4BD04_20230803163323.txt](https://github.com/free-dmo/free-dmo-stm32/files/12251927/E004010849D4BD04_20230803163323.txt)
@freepdk could you compile this for .bin file? It`s data tag from S0722530 label (1/2"x1" or 13x25mm)

queengooborg, 2023-09-15:
I've implemented both of these SKUs in a fork and compiled firmware accordingly, hope it helps! https://github.com/queengooborg/free-dmo-stm32/releases/tag/v1.0.8