free-dmo/free-dmo-stm32 #24: Need help setting up the board.

dylanstoel, 2023-01-02:
Hi there,

I'm pretty new to uploading the code to the bluepill and can't seem to find out how to do it, can someone help me with this?

Thanks in advance.

jbdunn82, 2023-03-23:
I am also having one heck of a time as the Hardware failed to mention you needed the serial to usb adapter. I have read where you can upload a bootloader so you can use the onboard usb micro but in order to dop that you have to use the adapter first. Ther eis also the jlink way and you have to have the adapter for that as well. Needless to say as a beginner on here to STM or Arduino whatever, the instructions are lacking. I just need someone to point me in the right direction.

KudzuKid, 2023-09-10:
I am having similar issues. This is a different beast than any Audrino I've played with.  I suspect Bluepill is the answer due to it's tiny footprint and installs easily in the device.  However, for those who are lacking firmware flashing procedures, can someone clarify / flesh out the process a bit?  

I  saw the note about needing to acquire a [cable ](https://www.amazon.com/dp/B07R8BQYW1?ref=ppx_yo2ov_dt_b_product_details&th=1) and I have that.  

Also, I've done the soldering, etc. Easy peasy (for me).  But the firmware side is a bit confusing in the article if you've never done this before. But otherwise, I'm entirely confused. I've tried a couple of YouTube videos but those seem equally ambiguous.  Some things say I need a USB dongle thing, but other articles suggest I don't.  FWIW, I have "STM32CubeIDE 1.13.1" - but have never used it - is it needed for this project?  PLEASE NOTE:  I don't intend to write my own (unless I absolutely had to!)  -  as I believe the images for the 2 main label types I use are already available.  Knowing that, how do I proceed?

Can anyone point us in the right direction?! Please and THANK YOU!

queengooborg, 2023-09-15:
The firmware flashing was something that had greatly confused me as well, so I just overhauled the setup documentation and wrote a detailed guide on how to flash the firmware on the bluepill!  https://github.com/queengooborg/free-dmo-stm32#install-the-firmware

Hope this helps!