free-dmo/free-dmo-stm32 #27: Does this actually allow unlimited prints?

MeisterLone, 2023-02-25:
As far as I can tell, when all the available UID's saved in the firmware are used up, the device wont allow printing anymore?

Or am I missing something

MeisterLone, 2023-02-27:
Anybody?

dtomcat, 2023-03-31:
D.MO RFID tag emulation

After startup of the printer a default tag emulation is used which can be defined in the firmware. The emulation will count down correctly until the end of the roll is reached. A power cycle / sleep+wakeup of the printer is enough to reset the emulated tag counter back to it's maximum.

If the original RFID board is also connected and a spool with D.MO RFID tag is found then the emulation data is updated with the data from this RFID tag. Just the counter will be emulated in this case. This enables the use of any D.MO format unless you have at least one original spool (you can peel the RFID label of that role and attach it to the inside of the printer).

MeisterLone, 2023-03-31:
Thank you, in the meantime I have tried it out. Very cool solution to the problem! I actually started my research into this device before knowing that this exists. I read the little spi chip that is on the device and found it to be entirely empty. I tried to dump the stm32 flash and it was readout protected. Next step was going to be fault introduction. Very clever solution you have here indeed. Did it take you a long time to decode what was beijg communicated by i2c?