free-dmo/free-dmo-stm32 #39: Wirigh connector

OGStyle, 2024-03-10:
Do you have a link to buy the female connector with wires on it? Thank you

KudzuKid, 2024-03-16:
For the **5XL** it appears to be a Molex 6 pin Picoblade, NOT a JST ZH, GH or SH, etc.   I can't speak to the other printers, but I wouldn't think (even) Dymo would change connectors across printers - they'd want to use a common board as much as possible I would think.

queengooborg, 2024-03-26:
I can confirm that the 550 also uses a Molex Picoblade connector.  I used the following product from Amazon for this project: https://a.co/d/9R8hogO

If you don't like/have Amazon, [Adafruit](https://www.adafruit.com/product/4926) is also a good choice.  You can also search for any of the following on your favorite stores: `6 pin Molex PicoBlade 1.25 mm` / `6pin 1.25mm MX Wire To Board Connector` / `JST MX 6pin`

(The instructions on the main branch incorrectly state the connector is a JST GH connector -- I have a [PR open](https://github.com/free-dmo/free-dmo-stm32/pull/34/files#diff-b335630551682c19a781afebcf4d07bf978fb1f8ac04c6bf87428ed5106870f5) to fix that!)

KudzuKid, 2024-03-26:
Yeah, I saw your PR yesterday, I think commented on it.

I used https://www.adafruit.com/product/4986 but didn't see your suggestion
until just now. Wish I saw it a couple months ago - that would be better
than my choice. Just chop that cable in half and you'd have both connectors
for the BluePill !

However, since the Dymo (5XL at least!) doesn't use the JST connector, the
https://www.amazon.com/dp/B07S18D3RN kit didn't do me any good (I bought it
or one extremely similar) off

Amazon when I thought it was a JST style.  By research and processor
elimination, I have since learned it's a Molex Picoblade (not a GH or SH or
ZH).

I appreciate your support and the Git PR - I am not familiar with GutHub
much at all - and this was my first real use of it and didn't know how to
submit those corrections other than making comments as I went along.

*The hack worked like a champ on my 5XL*.  The only thing I would
offer/suggest is maybe some electrical tape on the back of the 'Pill so it
doesn't accidentally short on any other components.  I've thought about
running a little extension of the USB port from the 'pill out the back to
avoid having to re-flashing the BluePill if I decide to change label types.
Any reason why that might not be practical? I'd hate to have to keep
removing those metal screws in the plastic chassis of the 5XL every time I
need to get at the 'Pill.

You're the only person I've seen active on this project lately, so glad to
chat with you.

Cheers!