free-dmo/free-dmo-stm32 #41: Resistor 4.7kΩ

forkyboy, 2024-03-16:
Hello,

I'm currently searching for a 4.7kΩ resistor, but I'm quite new to this and noticed there are various types of resistors available. Could you help me figure out the wattage I need for such a resistor? For instance, when I look on Amazon, there are many versions of a 4.7kΩ resistor, and I'd appreciate any guidance you could provide.

https://www.amazon.com/s?k=4.7k%CF%89+resistor&crid=2ORN48GS8L1RX&sprefix=Resistor+4.7k%CE%A9%2Caps%2C166&ref=nb_sb_ss_ts-doa-p_1_14

Also, I have a secondary question: Is Dymo implementing DRM to the extent that they take legal action against individuals who modify their printers? I'm curious because it seems odd that there isn't a market for modified Dymo 5XL printers, given this context.

Thank you very much for your help!

KudzuKid, 2024-03-16:
I suspect even the 1/4W would be fine, but use 1/2W if you just want peace of mine. Anything over what you really need isn't hurting anything, and usually gets more into issues of physical space & placement.   We're not dealing with a 1500W hairdryer or toaster oven here. These are low current, low-voltage circuitry here. 4.7K Ohm are extremely common.  As to WHY they are used, Let Me Google That For You... See here:
[](https://www.jotrin.com/technology/details/what-is-a-4k7-resistor-4-7k-ohm-resistor-color-code#:~:text=4.7k%20ohm%20resistors%20are%20commonly%20used%20in%20oscillator%20circuits,generate%20a%20precise%20frequency%20signal.)

forkyboy, 2024-03-16:
Thank you, everything is clear for my build now.

I appreciate all the infos.

<https://www.jotrin.com/technology/details/what-is-a-4k7-resistor-4-7k-ohm-resistor-color-code#:~:text=4.7k%20ohm%20resistors%20are%20commonly%20used%20in%20oscillator%20circuits,generate%20a%20precise%20frequency%20signal.>

KudzuKid, 2024-03-16:
> Thank you, everything is clear for my build now. I appreciate all the infos.
> […](#)

Updated with just another thought or two.