free-dmo/free-dmo-stm32 #42: DYMO LabelWriter 5XL Label Printer

DushanthaS, 2024-03-19:
Is this hack still working for DYMO LabelWriter 5XL Label Printer?

If so what 3rd party labels have you had success with?  Please let me know the search terms if you can't mention the brands or links here. 
Thanks!

KudzuKid, 2024-03-19:
Just buy generic labels that correspond to the Dymo equivalents (to the
binaries provided). Look on Amazon or wherever.

Is your 5XL not working with the hack?  Hmmm... Interesting. I haven't
heard anyone else having this issue, not sure that the problem might be.
Good luck.

DushanthaS, 2024-03-19:
ok Thanks for the quick reply, I just bought the printer but haven't tried the hack yet. I will make sure to report the progress here.