free-dmo/free-dmo-stm32 #43: RFID BOARD usage ?

forkyboy, 2024-04-06:
Hi everyone,

I've successfully completed the build, and I'm thrilled with the progress. My next step is to integrate an RFID board. My understanding is that this addition will enable compatibility with various paper sizes. However, I'm uncertain about the software integration:

What specific actions are required on the software side to support this feature?
Is there a particular binary file (bin) I need to include in the build to facilitate this? Do I need to incorporate all available binaries, or is there a specific one designed for managing multiple paper sizes?
I apologize if my questions seem basic; English is not my first language, and I'm trying to navigate this project with a limited understanding of the technical terminology. Could someone guide me on how to achieve compatibility with multiple paper sizes using the RFID board? Your assistance would be greatly appreciated. Thank you!

KudzuKid, 2024-04-06:
Hi There, 

I'm not sure if this will help or hurt. Rest assured, the _intent_ is to help

OK, so you say you've "completed the build" - I presume you mean you just completed the Blue Pill hookup? That's what the basis of my additional comments will be made on. 

So, based on your question, I'm thinking you haven't yet flashed (loaded) a .bin to the Blue Pill. You'll want to do that  after you complete your soldering.  In order to accomplish that, you'll need something like this:

[ST-Link V2 Emulator Downloader Programmer](https://www.amazon.com/gp/product/B07SQV6VLZ

It doesn't have to be this exact one, but something like it - there are others.  So one end of that (USB) plugs in to the host PC with the .bin on it. 

The other end gets connected to the Blue Pill. 

Which .bin do you need?  That depends!  :-)   Which label do you want to use in your printer?  Each .bin corresponds to a different label type.  There are shipping labels, address labels, "hello" name tag labels, etc.   The number of the .bin corresponds to the D*MO part #.  If you have an original D*mo box ("Authentic" or old school - doesn't matter) there's probably a part # on it. Look for (that part #)  + ".bin" and that's the one you want to flash the Pill with.  Or check Amazon, OFfice Depot, etc. They always show the part # - it's the default info needed to buy the correct label people need.  Does that make sense?  

Note: Changing label sizes / types is a bit annoying - you'll need to pull the screws and re-flash it with the appropriate .bin for the next label.  I'm pondering just "outboarding" the connector (or an extension on it, to be precise). 

Hope something here helps!

eha002, 2024-09-23:
I think i can help here. There is no software change for bypassing vs intercepting, if no NFC tag is present both just emulate a single Chip

With the intercept setup, on bootup, the Bluepill copies the tag data if present and emulates that tag instead.
It prevents the dymo from updating the chip so you only need 1 genuine roll of that size and can then go aftermarket and reuse them,
[image]

I'm developing a 3d printed tag holder currently to for easier hot swapping rolls.

-

A few trains of thought here on whats best for you, it will depend on Which printer you use and how often you swap barcodes, if you print direct or via dymo app, etc etc.

if you outline how you mainly use and how often you swap i can make more targeted recommendations

-

To break it down a lot, The NFC reader isn't that useful to you unless you:
Print more than 220 Labels at a time
Use Dymo App & Swap label sizes Regularly
Use Dymo App & Don’t like it complaining that the selected Labels are wrong.

another limitation is you cant print bigger paper than the Programmed Bin File.

so a board programmed as SD0904980 will allow you to print up to 220 Labels of any size that are 4XL compatible as long as you manually select the label you have installed in dymo, or correct paper size in Direct printing.

kestutiz, 2025-01-30:
Probably with little modification possible to program bluepill via USB. As I read for reprograming no need to move jumper in this case. Just upload bootloader and then program via USB. Not tested yet. If it works, then just make hole for USB to go outside the case.