free-dmo/free-dmo-stm32 #46: Data Label for 11356

fefed22, 2024-06-17:
Hello, 

anyone have data for SKU 11356 ?

Thanks in advance.

Best regards

KudzuKid, 2024-06-17:
Hi There,

YAMMV, but it appears that is also known as the S0722560.  Unfortunately,
that won't help.  I don't see that one in the .bin's either. Maybe you
could buy a roll and read the (NFC)  core (and share it!).  Just a
thought.  Best wishes.

Can anyone else help?

fefed22, 2024-06-19:
Hello,

buy a roll is not a problem, but do you have a helper to extract the NFC core ?

Best regards

fefed22, 2024-06-19:
i see this repo : https://github.com/free-dmo/free-dmo-tag-dump
i will test it and come back to you... :)

fefed22, 2024-06-19:
This app crashed everytime i want to launch it !
Do you have another one ?

fefed22, 2024-06-19:
i think that my phone have no NFC reader !

fefed22, 2024-06-20:
I've succeeded to read the NFC from the standard roll include with the machine. 
So i will test it on the real roll :)

KudzuKid, 2024-06-20:
Wishing you success...

fefed22, 2024-06-25:
Hello,

i've dump the source from roll.
i've installed ARM toolchain
i've put the source file at the root folder of this git archive.

But when i'm starting the make command i've an error : 
`Src/main.c: In function 'InitEmulationWithDefaultData':
Src/main.c:1021:31: error: 'DMO_TAG_SLIX2_BLOCKS' undeclared (first use in this function); did you mean 'EMU_SLIX2_BLOCKS'?
 1021 |   memcpy(EMU_SLIX2_BLOCKS,    DMO_TAG_SLIX2_BLOCKS, SLIX2_BLOCKS*sizeof(uint32_t));
      |                               ^~~~~~~~~~~~~~~~~~~~
      |                               EMU_SLIX2_BLOCKS
Src/main.c:1021:31: note: each undeclared identifier is reported only once for each function it appears in
make: *** [build/main.o] Error 1
`

Do you have any suggestion ?

Best regards

fefed22, 2024-06-26:
@KudzuKid i've attached the source file if you could compile file in your side.

Best regards 
[E00401085A6683B9_20240624123402.txt](https://github.com/user-attachments/files/15983784/E00401085A6683B9_20240624123402.txt)

KudzuKid, 2024-06-26:
I'm just a user here, I don't know if I can store anything in a repository,
etc.

fefed22, 2024-06-27:
Ooooh Sorry.
@brainiak4431 could you please see this thread ?!

fefed22, 2024-07-09:
Hello, @queengooborg can you look at this thread ?
i've attached the source file above.

Thanks

queengooborg, 2024-07-09:
Thanks for providing a dump file, I added it to my fork and just created a release, available here!  https://github.com/queengooborg/free-dmo-stm32/releases/tag/v1.0.13

fefed22, 2024-07-09:
Yeah ! Perfect.
i'll test it when i'm receive pin cable...

ChrisGru01, 2024-07-24:
> Thanks for providing a dump file, I added it to my fork and just created a release, available here!  https://github.com/queengooborg/free-dmo-stm32/releases/tag/v1.0.13

Can you explain to me exactly how to compile the bin? I keep getting the error as the user mentioned above.

fefed22, 2024-08-20:
Pin cable received, bluepill board installed and my printer works very well with my old label rolls !
Thanks for your help.

fefed22, 2024-08-20:
I close my issue.

closed 2024-08-20 by fefed22

ChrisGru01, 2024-08-20:
> Pin cable received, bluepill board installed and my printer works very well with my old label rolls !
> Thanks for your help.

Which printer do you use the bluepill with?

fefed22, 2024-08-20:
I'm using Dymo LW 550 ;)