free-dmo/free-dmo-stm32 #47: Error while trying to compile

ChrisGru01, 2024-07-24:
Hey, 

First of all, sorry for my bad English.

I wanted to compile my own .bin because it wasn't available for download. Unfortunately I encounter the following error with the "make" command. what do I have to declare there?:

[image]

I also wanted to ask how it works with the RFID chip. Can I then simply hold up the originals and it will recognize which label it should simulate?

queengooborg, 2024-07-24:
> Unfortunately I encounter the following error with the "make" command. what do I have to declare there?:

The `DMO_SKU` variable must be defined to specify which SKU you wish to compile firmware for (like `make DMO_SKU=30252`), but it looks like that wasn't documented anywhere.

> I also wanted to ask how it works with the RFID chip. Can I then simply hold up the originals and it will recognize which label it should simulate?

The bluepill will emulate whatever label you've flashed the firmware for, so if you've compiled and flashed the firmware for SKU 30252, that's the SKU that it will emulate.  However, if you've connected the RFID reader and the optional resistors, then if a real tag is detected, it will pass the data through.  Hope this explanation helps!

ChrisGru01, 2024-07-25:
> The `DMO_SKU` variable must be defined to specify which SKU you wish to compile firmware for (like `make DMO_SKU=30252`), but it looks like that wasn't documented anywhere.
>
>
> The bluepill will [...]

I actually did it by changing the main.c file and defining the SCU there. Is that also possible or do I have to do it in the make command?

queengooborg, 2024-07-25:
SKU 30321 has not been added to this repository, but I added it in on my own fork: https://github.com/queengooborg/free-dmo-stm32

From there, you can find a recompiled BIN for the SKU you're looking for too!

ChrisGru01, 2024-07-25:
> SKU 30321 has not been added to this repository, but I added it in on my own fork: https://github.com/queengooborg/free-dmo-stm32
> 
> From there, you can find a recompiled BIN for the SKU you're looking for too!

Thanks for compiling.

I'd just really like to figure out what I'm doing wrong and how to add more if necessary. Would it be possible for you to explain it to me? Currently I'm also getting this error:
[image]

queengooborg, 2024-07-25:
The issue there is a small one -- you just need to replace the underscore with an equals after "SKU" (so `DMO_SKU=30321` instead of `DMO_SKU_30321`), and you should be all good to go!

closed 2024-07-25 by ChrisGru01