free-dmo/free-dmo-stm32 #48: Frequent disconnects

EldoradoGit, 2024-08-28:
Hi There - i've modified a printer successfully, but it seems to have an issue where it requires frequent restarts of the printer (power light will randomly start flashing). It's happened 3-4 times a day, every day. I've opted to keep the rfid reader attached.

Do you have any recommendations on areas i should look?

KudzuKid, 2024-08-28:
Hi There,

Without knowing exactly the state of things, quality of workmanship, etc.
(no offense, please!), it's hard for me to say.

However, after some fair careers in IT and electronics, I might just make a
guess:

Cold/intermittent/shorting soldered connections?

Normally, as I understand it - the way my 5XL works - I only get the rapid
blink once I start Dymo Connect.  Once I press the power button again, it
behaves for the duration of the power cycle unless/until I quit the app or
power cycle again.

So it makes me think something is 'spoofing' the signal to check the
printer - and if it's happening randomly, then that might be either a bad
solder connection (as above) or perhaps corrupted firmware?  Just guessing.

Perhaps someone with more experience than I can help.

All the best!

EldoradoGit, 2024-08-29:
thanks for the comments! No offense taken, no worries!

I looked over the solder joints, and even reflowed them with air and flux. they look good to me.

one thing i noticed that i didn't notice before is that i dont think the board is genuine. The chip looks genuine when referring to guides online, but the reset button says "RESRT". I'm wondering if that's introducing some instability issues.

Another factor is our implementation is a little different. We are not printing from Dymo Connect, we're using Dymo's API to print directly from our software. I'm wondering if something is happening there...

EldoradoGit, 2024-08-29:
> Hi There, Without knowing exactly the state of things, quality of workmanship, etc. (no offense, please!), it's hard for me to say. However, after some fair careers in IT and electronics, I might [...]

I've got some more information.

I noticed that the Dymo will go to "sleep" (power LED gets dim) after 30 seconds if the DymoConnect software isn't running.
The stock/original Dymo will recover from this sleep state, but the modified Dymo will not, requiring a power cycle.

EldoradoGit, 2024-08-30:
And some more information.

With a modified 550 plugged in, and DymoConnect is NOT running, the printer will go to sleep after 30 seconds (USB logo LED on the front goes out).
If I insert an official Dymo roll, OR a generic roll, and then attempt to print the USB LED comes on and the power LED will begin to flash.
If I attempt to print while the USB logo is off and Power is flashing, i will get an "Unknown Label" error in the print queue.

After another 30 seconds the printer will go to sleep again. The USB LED will go out, and the flashing Power LED will stop flashing and go dim.

If I attempt to print again, it will print normally.

So it boils down to if the printer enters sleep from a connected state, it will not detect the label (genuine or generic), and will not print.
If the printer enters sleep from a disconnected/flashing state, it will detect the label and you can print.

Very odd...