free-dmo/free-dmo-stm32 #49: Make compatible with other microprocessors

queengooborg, 2024-09-25:
This is probably a far-fetched idea, but given how difficult it is to source genuine blue pills, it might make a lot more sense to make the project compatible with other microprocessors, like the Arduino Nano / Teensyduino.

This suggestion comes from absolutely no research on how difficult it would be to convert the existing codebase, or how challenging it would be to provide support.  My suggestion is purely based on supply chain issues with the STM32 blue pills!

MeisterLone, 2024-09-25:
They arent 'hard to source', i bought 3 on amazon for $3 each. Next day delivery.

queengooborg, 2024-09-25:
I'm talking about _genuine_ bluepills.  Counterfeits are so easy to obtain off of Amazon, eBay and anywhere else, but I have yet to find a seller that provides _genuine_ STM32F103 bluepills.  :)

MeisterLone, 2024-09-25:
See below amz link. 100% geniune chip made by ST
https://a.co/d/h4k4F8b

queengooborg, 2024-09-25:
That's not a compatible board; that's the C6, not the C8.  In the readme of the project: "Some places sell blue pills with STM32F103C6T6 which are not compatible."

MeisterLone, 2024-09-26:
True! My bad- I used a C8 from amz without problems though.

Mattyonemillion, 2026-04-17:
[image]

Working PoC with ESP32 over Wifi. Can add labels/remove/switch via GUI

BSwitchblade88, 2026-09-06:
@Mattyonemillion I've been trying to achieve a similar solution and using ESPhome as the base; that gives easy access to a web UI for sticker selection & changes.

Have you got your code uploaded somewhere already?